PyFixate/Fixate #1: Remove dependency on msvcrt so we can run on non-Windows environments

clint-lawrence, 2018-08-24:


clint-lawrence, 2018-08-25:
https://github.com/kcsaff/getkey & https://github.com/moses-palmer/pynput look like two different portable options that might help

kmolnar, 2018-08-27:
Resolved in version 0.3.

msvcrt removed. Now uses pynput module to monitor keyboard.

pazzarpj, 2019-01-31:
Reopened in light of #56

jcollins1983, 2019-02-08:
In addition to my previous comment, when running from the command line, with the git checkout installed via "pip install -e "path to setup.py", the error that is seen when running in PyCharm is not seen, however, the keypresses are not recognised with the exception of the enter key. Enter will advance the test, when numbers are pressed to enter the serial number nothing shows up.

The Pypi installed fixate has no issues.

Although it isn't clear from the image alone, I have attempted to enter a serial number in the terminal window on the right.

[image]