PyFixate/Fixate #11: fixate/reporting/csv.py Uses optional data to make directory structure

pazzarpj, 2018-09-26:
Currently, the csv file assumes that part_number and module are parsed to the program launch through --script-params.

Need to change so that the folder structure of the csv logs are able to be configured through a configuration file.

https://github.com/PyFixate/Fixate/blob/bb816960280feea263288c898ca360fbf4b7f34f/src/fixate/reporting/csv.py#L118

pazzarpj, 2018-09-26:
Also
https://github.com/PyFixate/Fixate/blob/bb816960280feea263288c898ca360fbf4b7f34f/src/fixate/__main__.py#L268-L275

clint-lawrence, 2018-09-28:
For now we could default to cwd and a timestamp (or timestamp-scripname) for the log. That would be a sensible default when testing/debugging scripts without requiring too many additional config hooks just yet

PR #18 (2018-10-05): Config file updates
Config file updates to resolve #11 and #12.

Added ability to have default config path as <python_env>\fixate.yml
Added -c to specify config files
Added some config helper functions for loading yaml, loading config parameters from plugins/core modules.
Changed the default csv parameters to work with default launch parameters with no config files.
closed 2018-10-05 by pazzarpj