PyFixate/Fixate #12: Modify how -c command operates and fixate.config

pazzarpj, 2018-09-26:
Currently, the -c command is used to parse in a yaml config file which is only used to set the csv path for reporting.

I am thinking we can use this as cumulative command to parse multiple config files via the command line.

This would allow arrangements where there is a default config you want set up for your environment but also want a specific set of parameters for this particular call out.

We should also consider parsing also having the ability to find a config file in the test scripts zip file.

pazzarpj, 2018-09-27:
I think this will help solve issue #11. In the context of that issue.
 
The idea is to parse in a config such as 
```yaml
part_number: 12345
plugins:
  fixate.reporting.csv:
    time_stamp_template: {0:%Y}{0:%m}{0:%d}-{0:%H}{0:%M}{0:%S}
    csv_path_template: 
      - C:\fixate_logs
      - {part_number}
      - {time_stamp_template}.csv
```
Support for setting up plug-ins with plug in specific data. 
Should give an output path of something like.
C:\fixate_logs\12345\20180927-111303.csv

clint-lawrence, 2018-10-02:
Instead of (or as well as) being able to specify multiple -c arguments, I think it would be useful to have a default config file location. Possibly %VIRTUAL_ENV%\fixate.yml.

Then when setting up a dedicated virtualenv for deployment (like we typically do) we can have some fixed global config which is tied to the installation. For more ad-hoc scenarios, the -c could be used.

[Thinking about our work deployments for a second, if there is too much tied up in the -c argument it would make it hard to run a script as it would in production without the launcher. On the other hand, having log template config decentralised isn't nice either]

I kind off like the idea of looking in cwd and/or a script zip file too, but think we should not implement that until we come up with a real use case. At the moment, we wouldn't really want a script to be able to override the log setup for example. If a that changes, happy to consider the change then.

clint-lawrence, 2018-10-02:
For a bit of inspiration, pip explains the config options nicely in it's documents (but really more flexible that we need to worry about for now)
https://pip.pypa.io/en/stable/user_guide/#configuration

pazzarpj, 2018-10-03:
Following a discussion with Clint, I think the yaml file should be flatter to make it easier to specify overriding configuration files without accidently overriding. For example what I originally proposed.
```yaml
plugins:
  fixate.reporting.csv:
    time_stamp_template: {0:%Y}{0:%m}{0:%d}-{0:%H}{0:%M}{0:%S}
    csv_path_template: 
      - C:\fixate_logs
      - {part_number}
      - {time_stamp_template}.csv
```
would mean that a second yaml file
```yaml
plugins:
  my.custom.gui:
    param1: hello
    param2: world
```
Would remove original fixate.reporting.csv plugin from the config which is not the desired behaviour.

by changing it to something like:
```yaml
plg_csv:
  import_name: fixate.reporting.csv:
  time_stamp_template: {0:%Y}{0:%m}{0:%d}-{0:%H}{0:%M}{0:%S}
  csv_path_template: 
    - C:\fixate_logs
    - {part_number}
    - {time_stamp_template}.csv
```
would mean that a second yaml file
```yaml
plg_my_custom_gui:
  import_name: my.custom.gui:
  param1: hello
  param2: world
```
Would not override the previous config unless specifically added:
eg. 
```yaml
plg_csv:
  import_name: fixate.reporting.csv:
  time_stamp_template: {0:%Y}{0:%m}{0:%d}-{0:%H}{0:%M}{0:%S}
  csv_path_template: 
    - C:\my_custom_loc
    - {index}
    - {time_stamp_template}.csv
plg_my_custom_gui:
  import_name: my.custom.gui:
  param1: hello
  param2: world
```

pazzarpj, 2018-10-03:
As for a default configuration file location %VIRTUAL_ENV%\fixate.yml does not always exist, even in virtual environments unless the environment is explicitly activated.
In our current use case, the config can be seen at several levels.
Site config: Configuration specific to the company/site deploying. Things contained would be things like details for connecting to databases for reporting.
Workstation: Instrumentation configuration, serial port locations etc.
Dev/Testing: Parse in a config to override default settings and behaviour
Script specific: Configuration which needs to be shipped with a specific config.

I think we can easily do:
- Site config: root directory of python executing (<venv_path>/fixate.yml)
- Dev: -c parameter
- Testing: -c parameter or override parameters in fixate.config

Still supports but clunky
- Workstation: -c
- Script specific: -c

closed 2018-10-05 by pazzarpj