PyFixate/Fixate #17: Logging of user wait time is inaccurate

pazzarpj, 2018-10-04:
default reporting.csv is set up to log the time when the system is awaiting a user response.
This implementation is insufficient for two reasons.

-  We don't get a representation of the start and end of the user wait so we can't know how long we waited
- The hook used relies on pub sub which means that blocking actions will delay the stopwatch (eg. cmdline blocks until user response so the wait time is actually reported when the user response is complete whereas the gui is non-blocking and will log when the user response is started.

clint-lawrence, 2018-10-04:
I think there are two possible approaches.

1. Have the sequencer to fire a log prior to any blocking call and immediately on the calls return. That gives us two logs with timestamps (we could call them for example, "user waiting" and "user finished" logs)
2. Have the sequencer record the start time when launching a blocking call. Then after the blocking call returns, calculate the duration of the blocked call and log that value.

pazzarpj, 2018-10-10:
Log seems to correctly report user wait start and end times according to launching tiny_images.py
[image]

closed 2018-10-16 by pazzarpj