PyFixate/Fixate #20: Exit code is not 0 on sucess

clint-lawrence, 2018-10-04:
I think the problem is from __main__.FixateWorker.ui_run. Pretty sure we just need to change the 5 to a 0, but it seems too easy.. Your thoughts, @pazzarpj?

`        finally:
            unregister_csv()
            if serial_number == "ABORT_FORCE" or test_selector == "ABORT_FORCE":
                return 11
            save_local_config()
            self.clean = True  # Let the supervisor know that the program is finishing normally
            if self.sequencer.end_status == "FAILED":
                return 10
            elif self.sequencer.status == "Aborted":
                return 11
            elif self.sequencer.end_status == "ERROR":
                return 12
            else:
                return 0
`

pazzarpj, 2018-10-04:
I think Kieran changed it from exit code 0 to 5 on success because qt will return exit code 0 on some scenarios of force exit. This means we were getting incorrect pass return codes when GUI crashed.

clint-lawrence, 2018-10-04:
I assume our work deployment knows that and accommodates for it. Not nice, but possibly no easy fix either.

pazzarpj, 2018-10-04:
Yes it currently is looking for exit code 5.
From my short exposure with PyQt, there are some perculiarities such as it overriding the default python exception hook and so forth. There is probably a way or event to catch any unexpected exit code 0s but that's for another day.

clint-lawrence, 2018-10-04:
Agree. I'm going to close this for now. While it's unconventional, reliable is more important for this, I think.

closed 2018-10-04 by clint-lawrence