PyFixate/Fixate #54: User action causes the next user command to hang GUI

pazzarpj, 2018-11-26:
example

```python
def my_test():
    return True

user_ok("This displays")
user_action("Display This", my_test)
user_ok("This hangs the GUI")
```

pazzarpj, 2018-11-27:
[image]

Log from Autocard test jig ran in GUI.

Seems to cause a similar issue at the end of the sequence which prompts test finished. Will cause the exit code to show error even though the test was successful.

clint-lawrence, 2019-05-15:
I can't replicate on master. The full test script I'm using is: 
```
"""
This is a test script that shows basic use case for the fixate library
"""
from fixate.core.common import TestClass
from fixate.core.ui import user_action, user_ok

__version__ = '0'

class ATest(TestClass):
    """
    Asks the user to push the red button
    """
    test_desc = "A Test"

    def test(self):
        class MyTest:
            def __init__(self):
                self.count = 0

            def my_test(self):
                user_ok(str(self.count))
                self.count += 1
                return self.count > 3

        c = MyTest()
        user_ok("ok: This displays")
        user_action("action: Display This", c.my_test)
        user_action("action: Display This", lambda: True)
        user_ok("ok: This hangs the GUI")

TEST_SEQUENCE = [ATest(), ATest()]
```

@pazzarpj, any thoughts on how to replicate?

pazzarpj, 2019-05-15:
My original code block I think used to cause the issue. I'm a bit out of action to check others at the moment

clint-lawrence, 2019-05-15:
@pazzarpj will add more info soon, but it's looking like a dependency issue

clint-lawrence, 2019-05-16:
When I couldn't replicate the failure above, there was a bug in my code. I'd started clean up some things in user_action. But in doing so, I made a type and was sending message topic "UI_**A**ction", instead of "UI_**a**ction". The correct topic is with a lower case "a".

So after fixing that issue, I can replicate the bug as described. In thinking about the problem, I've
also come to the conclusion that there are two fairly distinct usages of user_action. We may or may not be able to accommodate both when fixing the bug. The target function supplied to user_action can do two different things:
1.  Interact with the user with other function call. e.g. calling user_ok or user_yes_no. I'll have to review some places where this is used, but I'm not sure we even need a user_action in this case. We could just use a loop in the test script.
2.  Don't interact directly with the user, but need a way of user to be able to cancel (and most likely mark that as a fail)

In the first case, the test must ensure there is a valid escape sequence. We could possibly look further at that to see if there is a case to make a default 'escape'. (I would like to see that anyway, even to stop a long running test and exit the sequence without hitting the "X")

So, using the script below and 0.5.1 with the GUI, the first test hangs after clicking 'continue' on the first "user_ok" call in the target. The second test hangs the GUI if left to timeout. If the user_action is cancelled by hitting "FAIL" the sequence ends without hanging, but is marked as a "PASS", since calls to check functions need to be explicit.

Using the script below and 0.5.1 with the command line, nothing hangs. However the second test can't be cancelled. We should be able to cancel by hitting ESC, but that doesn't work.
```
"""
This is a test script that shows basic use case for the fixate library
"""
from fixate.core.common import TestClass
from fixate.core.ui import user_action, user_ok

__version__ = '0'

class InteractiveUserActionTest(TestClass):
    """
    As the user to hit O.K. inside the user_action
    """
    test_desc = "InteractiveUserActionTest"

    def test(self):
        class MyTest:
            def __init__(self):
                self.count = 0

            def my_test(self):
                user_ok("ok in action target: hit enter to continue " + str(self.count))
                self.count += 1
                return self.count > 3

        c = MyTest()
        user_ok("ok: About to call user_action")
        user_action("action: Display This", c.my_test)
        user_ok("ok: user_action done")

class NonInteractiveUserActionTest(TestClass):
    """
    user_action will timeout, but the use can FAIL early.
    """
    test_desc = "NonInteractiveUserActionTest"

    def test(self):
        import time

        class MyTest:
            def __init__(self):
                self.count = 0

            def my_test(self):
                self.count += 1
                time.sleep(0.5)
                return self.count > 10

        c = MyTest()
        user_action("wait 5 s. or press FAIL to cancel", c.my_test)

TEST_SEQUENCE = [
    # InteractiveUserActionTest(),
    NonInteractiveUserActionTest()
]
```

closed 2019-05-22 by clint-lawrence