PyFixate/Fixate #55: Document instruction to use ftdi on linux

clint-lawrence, 2018-12-04:
From a helpful user:
[[[
Install stretch image on the pi //(Earlier images don’t support python 3.5)

sudo apt-get update
sudo apt-get upgrade

sudo apt-get install python-serial
sudo apt-get install libftdi-dev
sudo apt-get install python3-pip
sudo pip3 install -U pip setuptools wheel
sudo pip3 install pylibftdi

// Edit the USB rules file
sudo vi /etc/udev/rules.d/99-com.rules

// Add these lines to the file
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", GROUP="dialout", MODE="0660"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6014", GROUP="dialout", MODE="0660"

// Reload rules after edit.
sudo udevadm control --reload-rules

//test USB driver to see if present. (The USB device needs to be plugged in after the previous command but before this one..)
python3 -m pylibftdi.examples.list_devices

sudo pip3 install ruamel.yaml
sudo pip3 install --upgrade six
sudo pip3 install fixate
]]]

pazzarpj, 2019-01-31:
#77 Updated the ftdi driver to use between linux and windows.
Installing the .so file in linux is also in the Exception message if not correctly installed.
@clint-lawrence is this instruction still required?

clint-lawrence, 2019-02-01:
I'm happy to close this now. I've created #79 with a suggestion to improve #77.

closed 2019-02-01 by clint-lawrence