PyFixate/Fixate #56: Fix gui imports so fixate runs on a headless Rpi

clint-lawrence, 2018-12-04:
fixate should be able to run without the gui. That includes not importing gui libraries and having dependencies on X. When trying to run on a headless Rpi fixate was failing without having an X server running.

clint-lawrence, 2018-12-27:
I think the problem is in the pynput module. This is used to capture keyboard events.
```
File "/home/pi/fixate-venv/lib/python3.5/site-packages/fixate/ui_cmdline/cmd_line.py", line 7, in <module>
    from pynput import keyboard
File "/home/pi/fixate-venv/lib/python3.5/site-packages/pynput/__init__.py", line 23, in <module>
```

pazzarpj, 2018-12-27:
Cool. We can move it to requirements_test.txt

pazzarpj, 2018-12-27:
Actually, I think that was done by Kieran for the keyboard hooks. Might just need to find another cross platform solution.

clint-lawrence, 2018-12-28:
Yep, exactly. Thinking it might just be easiest to switch back to termios for os x/linux and msvcrt on windows.

clint-lawrence, 2018-12-28:
This might work. Uses termios/msvcrt.
https://github.com/magmax/python-readchar

smontanari1, 2019-01-30:
Readchar is suitable alternative to pynput or msvcrt but does not have a function for testing if a key is being pressed, only recording the next keystroke and will block until you press a key. Need help to examine if possible to implement

closed 2019-02-18 by pazzarpj