PyFixate/Fixate #78: Clean up Virtual Mux

pazzarpj, 2019-01-31:
Linked to #75.
Virtual Muxes are double handling data by turning pin values into a bit stream and then back to pin values when it pushes it up to VirtualAddressMap via callbacks.
Suggest removing this intermediate step to clean up code and make it easier to follow the Mux Logic.

clint-lawrence, 2019-02-02:
Thinking out loud.
- Public API should stay the same; But we need to make it clear what that is.
- Some of the existing scant tests use the internal bit representation that we want to remove. A first step will be transforming those tests into something more suitable to check for regressions.

VirtualAddressMap.install_multiplexer shouldn't directly assign any attributes to the mux. But O.K. for it to directly access the pin_list.

I'm thinking VirtualAddressMap will have method named
set_pins_from_virtual_mux(new_pins, trigger_update=True, clear_pins=None, clear_time=0)
new_pins & clear_pins: collection of pin, value tuples. Value is True of False.
trigger_update is boolean
clear_time is in seconds

VirtualMux will then have a reference to the VirtualAddressMap, stored in an attribute, self.parent_virutual_address_map. Then in the multiplex method, we can directly call self.parent_virtual_address_map.set_pins_from_virtual_mux

Hopefully we can manage all this without effecting any existing jig definitions.

clint-lawrence, 2019-02-04:
Have started working on this here:
https://github.com/clint-lawrence/Fixate/tree/virtual-mux-refactor

closed 2024-06-28 by clint-lawrence