PyFixate/Fixate #79: hard coded path for libftd2xx.so won't work everywhere

clint-lawrence, 2019-02-01:
PR #77 added linux support for the existing ftdi driver for linux. The path to the shared library file is hard coded as '/usr/local/lib/libftd2xx.so'. That will break on installations that use a different install prefix.

Not sure if it's the right solution, but ctypes has a find_library function
https://docs.python.org/3/library/ctypes.html#finding-shared-libraries

@smontanari1, do you want to see if that works?

pazzarpj, 2019-02-01:
I belive @smontanari1 confirmed that the hard coded path was the one recommended by the ftdi docs. Although a more generic solution would be appreciated.

clint-lawrence, 2019-02-02:
Wonder if it's even possible to install the library elsewhere...? Not a big issue. Not sure I have anything FTDI related at home. Might bring something home and check on OS X also

pazzarpj, 2019-02-13:
I think I'll close this issue unless we can see a specific case of the library being in a different location.

closed 2019-02-13 by pazzarpj