PyFixate/Fixate #99: Handle user_choices for more than 3 options

pazzarpj, 2019-04-02:
Currently, user_choices allows an arbitrary amount of choices to be parsed to it.
However the UI does not handle this gracefully.

Separate out the common functions such as user_yes_no, user_ok, user_retry_abort_fail etc. into their own api and have the user_choices have a different way of selecting options.

clint-lawrence, 2019-05-19:
Have to got a use case in mind for user_choices?

Assuming #108 gets merged, the won't be a user_choices anymore, so we will be able to just close this issue :)

pazzarpj, 2019-05-19:
A current use case would be the natabase. Selecting location, gas bottles etc have an arbitrary amount of options to select. For the GUI we had to re-implement this use case with user input and manually verify the inputs.

clint-lawrence, 2019-05-19:
How many possible entries do you think we need to practically support? Up to a dozen or so is probably practical with a fairly simple UI.

So thinking about possible alternatives that are more universal going forward.
For command line:
print a numbered list of choices. The user would type the number & hit enter to make a selection. The number entered would be checked to make sure it was in range then return.
If the user hits ESC an exception would be raised, failing/erroring the test and allowed a way to exit (or retry).

For GUI:
Open a modal dialog (or use the image panel?). Display a list of radio buttons with and O.K. and cancel button. O.K. would return the selected choice and cancel would raise and exception, similar to cli ESC. Possibly at some point it could switch from radio buttons in a grid to a list box.