PyFixate/Fixate #103: Test script does not work as described in quick start guide

noplacebo, 2019-05-03:
The test script as stands requires the index to be set in the run command (ie "--index passes"). Without this the index defaults to "default" which is not set in examples/test_script.py

The result is a KeyError with the following traceback:
```
Traceback (most recent call last):
  File "G:\env\fixate0.5.0\lib\site-packages\fixate\__main__.py", line 309, in retrieve_test_data
    sequence = data[index]
KeyError: 'default'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "G:\env\fixate0.5.0\lib\site-packages\fixate\__main__.py", line 247, in ui_run
    test_data = retrieve_test_data(test_suite, self.args.index)
  File "G:\env\fixate0.5.0\lib\site-packages\fixate\__main__.py", line 311, in retrieve_test_data
    raise ValueError("Invalid test index parsed: {}".format(index)) from e
ValueError: Invalid test index parsed: default
```

pazzarpj, 2019-05-16:
@clint-lawrence Is this what you fixed?

clint-lawrence, 2019-05-19:
Yep, this will be fixed when #108 is merged

clint-lawrence, 2019-05-22:
#108 has been merged, which included this fix

closed 2019-05-22 by clint-lawrence