PyFixate/Fixate #128: Move the log to a better location

clint-lawrence, 2020-09-21:
The python logs are currently written to the current working directory. This should change to a better default location (appdirs.site_data_dir("Fixate") or appdirs.user_log_dir("Fixate")) Also, we would ideally add some configuration for the logging. If nothing else, a command line parameter to set the log directly.

jcollins1983, 2022-10-25:
@clint-lawrence , @John2202W's changes that were merged under PR #158 address the log location. Did you want to keep this issue open for the last 2 sentences?
Did you have any particular configuration in mind? The command line parameter is easy enough.

clint-lawrence, 2022-10-25:
We should at least check the tests to ensure they aren't dumping a bunch of logs to outside of the test folder when running locally. (similar to the issue @John2202W is fixing with https://github.com/PyFixate/Fixate/pull/160). We can possible do that by adding the the --disable-logs flag to the script2log tests.

John2202W, 2022-10-26:
Good point, have disabled the logs in #160

John2202W, 2022-10-26:
Do we still want a command line arg to set the name/location of the fixate.log 's ?

clint-lawrence, 2022-10-26:
Unless you're keen, I'd say just leave it until we have the need.

clint-lawrence, 2024-07-26:
fixed in #203 and #155

closed 2024-07-26 by clint-lawrence