PyFixate/Fixate #204: Documentation content updates

josh-marshall-amp, 2024-07-24:
After merging #194, the next step is the _content_ of the documentation. This issue raised to collect the updates we'll need to make.

Any discussion of the docs philosophy and howto from the [docs readme](https://github.com/PyFixate/Fixate/blob/master/docs/readme.md) can go here too.

- [ ] reorganise hierarchy for the docs (see the readme on doc quadrants)
- [ ] (not strictly docs) reorganise the fixate source to make clear what's public vs private
- [ ] fix up all the docstrings that don't render correctly
- [ ] add missing docstrings on important modules
- [ ] fix, then add a stack more docstrings, at least enough to cover the basics and provide an example
- [ ] `test_data` dict vs `TEST_SEQUENCE` [comment](https://github.com/PyFixate/Fixate/pull/194#discussion_r1685700632)

Docs to move into this repo (either as .rst or as docstrings in code, prefer the latter):
- [ ] [test script software structure](https://ampcontrol.atlassian.net/wiki/spaces/CPEK/pages/465043463/Test+script+software+structure?focusedCommentId=503218279) (Confluence): put in the top level just after the tutorial
- [ ] [Using Fixate Drivers in Python](https://ampcontrol.atlassian.net/wiki/spaces/CPEK/pages/445382657/Using+Fixate+Drivers+in+Python): should be in module+class+function docstrings
- [ ] [Jig driver migration](https://ampcontrol.atlassian.net/wiki/spaces/CPEK/pages/462356484/New+Jig+Driver+Migration+Guide)

clint-lawrence, 2024-07-24:
There's a discussion to be had about if we keep this repo public or not. If we keep it public, then we need be mindful of what is documented where. For example test script structure is not specific to fixate. Just how to setup our scripts internally and so should remain internal.

I'm also going to edit the description to add in a list of the most import modules to document.

josh-marshall-amp, 2024-07-31:
While reviewing a docs PR, occurred to me I really like when docs link to the source... that way it's easy to look at where something is missing or went wrong, particularly in this early stage of getting the docs fixed up.

[`sphinx.ext.linkcode`](https://www.sphinx-doc.org/en/master/usage/extensions/linkcode.html#module-sphinx.ext.linkcode) would let us link to Github, which would be cool if we could quickly fix something in the webeditor

[`sphinx.ext.viewcode`](https://www.sphinx-doc.org/en/master/usage/extensions/viewcode.html) renders the code itself as an extra set of HTML pages. Good but not as useful as linking to Github.

[This issue](https://github.com/readthedocs/sphinx-autoapi/issues/202) shows the process of linking to github... but right to the line number.

josh-marshall-amp, 2024-07-31:
[This VSCode extension](https://marketplace.visualstudio.com/items?itemName=njpwerner.autodocstring) will be useful for popping in a docstring template

josh-marshall-amp, 2024-08-02:
> Or you could just use PyCharm which does that out of the box ;)

If we're gonna do IDE battles 🔥 here... then if you want to rewrite my VSCode launch setup, which fires up the simulator with debugger attached to the Python side of the comms code in one pane and GDB attached to the executable's equivalent in C in another... I'm listening. 😉